elmot/clion-embedded-arm #7: Make sure `waitFor` is never called on UI thread

amakeev, 2017-10-23:
https://github.com/elmot/cubemx-clion/blob/ac7a2377e889ddb805a841b4a141e7354d9c832c/src/xyz/elmot/clion/openocd/OpenOcdRun.java#L54

elmot, 2017-10-23:
How can I delay debugger start for a while?

amakeev, 2017-10-24:
Since  `OpenOcdLauncher` needs to stop the current process in `createProcess`, you have 2 options:
1) Show a progress dialog to let the user know that IDE is waiting for something, with 'Cancel' button.
(`com.intellij.openapi.progress.ProgressManager#runProcessWithProgressSynchronously`)
2) Create your own `ProcessHandler` that will make sure only one backing `Ocd` process is present, doing this in background (more complicated).

closed 2017-10-29 by elmot