elmot/clion-embedded-arm #57: Detect Platform

yohboy, 2017-12-04:
## Expected Behavior
Can the plugin detect automatically the platform (Linux or Windows) and set default platform path for OpenOCD ? Don't know if a standard path exist for OpenOCD installation on Windows.

## Actual Behavior
Default to Linux (/usr)

## Specifications
  - Plugin Version:  0.8 alpha
  - CLion version: 2017.3
  - Platform: Windows

yohboy, 2017-12-05:
By default, if CLion doesn't find the path, or no path set, the default is : "C:\Users\Name" with Name the user session name. Could be a good solution. What do you think ?

elmot, 2017-12-05:
I personally do not keep stuff like tools in my home folder. On the other
hand, home folder always exists, a user always has write access to it... It
may be a better default rather than C:\

PR #58 (2017-12-07): Better default for OpenOCD home under Windows
Fixes #57
commit 2f49970f (2017-12-07): Better default for OpenOCD home under Windows (#58)

Fixes #57
closed 2017-12-07 by elmot