elmot/clion-embedded-arm #59: Cannot point to OpenOCD home on Ubuntu 16.04 Desktop  and CLion 2017.3

belussi, 2017-12-14:
## Expected Behavior
User can point to OpenOCD home directory on Linux

## Actual Behavior
Getting "OpenOCD Home is invalid" when pointing to OpenOCD home.

## Steps to Reproduce the Problem
  1. Create OpenOCD configuration for new project.

## Specifications

  - Plugin Version: 0.8.alpha
  - CLion version: 2017.3
  - OpenOCD version: 0.10.0
  - GDB Version: n/a
  - STM32CubeMX version: n/a
  - Toolchain: n/a
  - Platform: Ubuntu 16.04 Desktop
  - Hardware configuration:

elmot, 2017-12-14:
OpenOCD requires both `bin` and `share/openocd/scripts` folders in the structure. Most probably your OpenOCD setup is not complete. How does it look like on my Windows:
```
+---bin
+---drivers
|   +---FTDI CDM v2.08.28 Certified
|   |   +---amd64
|   |   +---i386
|   |   \---Static
|   |       +---amd64
|   |       \---i386
|   +---ST-Link
|   |   +---amd64
|   |   \---x86
|   +---WinUSB
|   |   +---amd64
|   |   \---x86
|   \---WinUSB_Mini
+---QuickSetup
\---share
    \---openocd
        +---contrib
        |   +---libdcc
        |   +---loaders
        |   |   +---checksum
        |   |   +---debug
        |   |   |   \---xscale
        |   |   +---erase_check
        |   |   +---flash
        |   |   |   +---at91sam7x
        |   |   |   +---fm4
        |   |   |   +---fpga
        |   |   |   +---kinetis
        |   |   |   +---kinetis_ke
        |   |   |   \---xmc1xxx
        |   |   \---watchdog
        |   +---remote_bitbang
        |   +---rpc_examples
        |   +---rtos-helpers
        |   \---xsvf_tools
        +---OpenULINK
        \---scripts
            +---board
            +---chip
            |   +---atmel
            |   |   \---at91
            |   +---st
            |   |   +---spear
            |   |   \---stm32
            |   \---ti
            |       \---lm3s
            +---cpld
            +---cpu
            |   \---arm
            +---fpga
            +---interface
            |   \---ftdi
            +---target
            +---test
            \---tools
```
On *nix it should look very similar way

belussi, 2017-12-14:
Hi,
Thanks for your reply.
I'm using standard ARM-MCU-Eclipse plugin installation and I see that 'scripts' directory is located directly under OpenOCD home directory. There is no 'share' at all.

`kamil@kamil-Latitude-E7450:~/workspace/same70-base$ ls /opt/gnu-mcu-eclipse/openocd/0.10.0-5-20171110-1117/
bin  contrib  doc  gnu-mcu-eclipse  info  licenses  man  OpenULINK  README.md  scripts
`

I have created it manually and it worked.

elmot, 2017-12-14:
Sorry, my plugin does not support that (yet?). Since you use ubuntu, try to install standard openocd distribution via `sudo apt-get install openocd`

elmot, 2017-12-14:
@belussi, looks like you are trying to use this stuff with Microchip MCUs. Please let me know if you gain any success or not.

belussi, 2017-12-14:
Flashing seems to working fine. But I get this when trying to debug:
`Error during pretty printers setup: Undefined info command: "pretty-printer".  Try "help info".

Some features and performance optimizations will not be available.

info pretty-printer
Undefined info command: "pretty-printer".  Try "help info".

Debugger connected to tcp:localhost:3333
`

elmot, 2017-12-14:
Looks like you are using very old gdb. Try version 8 or at least 7.

belussi, 2017-12-18:
Hi,

My SAME70 seems to be working fine with your plugin.

mintisan, 2018-01-02:
It works for me with the directory you mentioned, thanks. @elmot 

I used openocd binary from https://github.com/runtimeco/openocd-binaries

elmot, 2018-01-04:
I got several requests to fix this through private channels.

mintisan, 2018-01-05:
What else do you need except for openocd binary file? Why need the structure completely?

elmot, 2018-01-05:
Bunch of scripts.

elmot, 2018-01-05:
Bunch of scripts.

keatis, 2018-01-09:
Some unofficial packages do not have `share/` folder, but have `scripts/`right at top level.

As a workaround on Windows, just copy whole `scripts/` folder to `OPENOCD_HOME/share/openocd`, so that there's `OPENOCD_HOME/share/openocd/scripts/board` folder. Or make a link (i didn't try this).

On *nix just make a symlink with smth like
```bash
cd OPENOCD_HOME && mkdir -p share/openocd && cd $_ && ln -s ../../scripts
```

PR #72 (2018-01-10): Smart script search
Fixes #59
closed 2018-01-10 by elmot