elmot/clion-embedded-arm #64: Feature request: Pass user OpenOCD arguments

kiciek, 2017-12-20:
Right now all controll user has over OpenOCD is selecting board script
1. Add freedom to chose more then one script (separate target and interface scripts)
2. Change OpenOCD commands from default `;reset;shutdown` to any.

AC6 System Workbench 4 STM32 has interface with many options
[image]
[image]

elmot, 2017-12-20:
1. I decided not to do this. Feel free to write your own board config and place it, for instance, into your project - the syntax is pretty trivial. I think I need add that to the documentation.
2. Good point.

mchodzikiewicz, 2018-01-03:
I am also very interested in those features.
I would like to point out, that 'board' approach is only valid as long as you use devkit with debugger onboard, when you use custom hardware, it would be preferred not to rewrite config each time I change my probe, since probe and target are separate devices.

kiciek, 2018-03-20:
Sorry, but I cannot find it myself. How do I put verify in script to check if firmware was loaded without errors?

elmot, 2018-03-28:
@mchodzikiewicz , you may use your on per-project board config file. Those files in most cases contain just two includes -for the probe and for the target.

PR #101 (2018-05-03): Custom debug run
Fixes #75 
Fixes #64 
Fixes #24

---
This change is [[image]](https://reviewable.io/reviews/elmot/clion-embedded-arm/101)
closed 2018-05-03 by elmot