elmot/clion-embedded-arm #65: Feature request: Move board selection to Run/Debug configuration

kiciek, 2017-12-20:
Move `Board Config File` and any future OpenOCD configuration from `File->Settings->Build, Execution, Deployment->OpenOCD Support` to `Run/Debug Configurations->OpenOCD Download & Run` to allow fast switching between targets or interfaces or both.

elmot, 2017-12-20:
Is that really helpful? The plugin config is specific to the project, i hardly can imagine what for that fast switch is done w

kiciek, 2017-12-21:
It is useful to run with and without semihosting, or to debug running firmware without resetting target.
Sadly both configurations are saved in .idea/workspace.xml and because of that are not possible to share by version control.

makkarpov, 2018-02-10:
It's also useful when you have to debug multiple boards at once, so you can have different board scripts to select individual boards.

+1 for feature

PR #99 (2018-04-08): Move run board name and ports settings to the run configuration
Resolves #65

---
This change is [[image]](https://reviewable.io/reviews/elmot/clion-embedded-arm/99)
commit edbe4ffa (2018-04-08): Move run board name and ports settings to the run configuration (#99)

Resolves #65
closed 2018-04-08 by elmot