elmot/clion-embedded-arm #66: Linker issue with DSP example Code

aul12, 2017-12-26:
I generated a makefile using the plugin and a project for the STM32-L432KC. There are a lot of linker problems regarding the Drivers/CMSIS/DSP_Lib/Examples Folder, as far as i can see the examples are
all standalone programms. Is it possible to exclude this folder from the CMake by default?

## Specifications

  - Plugin Version: 1.0.beta1
  - CLion version: 2017.3
  - STM32CubeMX version:4.23.0
  - Toolchain: arm-none-eabi-gcc
  - Platform: ubuntu 16.04-arm64

elmot, 2017-12-26:
Looks like you are using old firmware package in Cube. Try to update it to very last one (1.10.0) via Cube's menu  _Help ->Check for Updates_

aul12, 2017-12-28:
Thanks for the help!

closed 2017-12-28 by aul12