elmot/clion-embedded-arm #70: not in execute format: File Format not recognized when running debug

mintisan, 2018-01-02:
## Expected Behavior

Debug normally.

## Actual Behavior

Shows `Firmware Download Success`, but `File Format not recognized` as below:
[image]

And my gdb and openocd seems fine as below:

```
➜  openocd-bin-0.10.0-MacOS file /usr/local/bin/arm-none-eabi-gdb
/usr/local/bin/arm-none-eabi-gdb: Mach-O 64-bit executable x86_64
➜  openocd-bin-0.10.0-MacOS file local/bin/openocd
local/bin/openocd: Mach-O 64-bit executable x86_64
```

If I missed any settings ?

## Steps to Reproduce the Problem

  1. Run -> Build
  1. Set Breakpoint.
  1. Run -> Run OCD f4-blog : Firmware Download Success
  1. Run -> Debug OCD f4-blog
[image]
[image]

## Specifications

  - Plugin Version: 1.0.beta1
  - CLion version: 2017.3.1
  - OpenOCD version: 0.10.0
  - GDB Version: GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
  - STM32CubeMX version: 4.23.0
  - Toolchain: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
  - Platform: OS X Captain 10.11.6 
  - Hardware configuration:

elmot, 2018-01-02:
Wrong GDB is used. Unfortunately, Mac version of CLion is shipped with single-platform gdb now.
Please uncheck "GDB shipped with CLion" at 1st shown screen and try again.

closed 2018-01-02 by elmot

mintisan, 2018-01-03:
@elmot It works for me. Thanks.

[image]