elmot/clion-embedded-arm #80: What if spaces and intl characters are in the project path

elmot, 2018-01-31:
Check and fix if broken

blaisejarrett, 2018-02-25:
This appears to be a problem on projects with spaces in their path under windows.
While flashing:
`"C:\Program Files\openocd-0.10.0\bin\openocd.exe" -c "tcl_port disabled" -s "C:\Program Files\openocd-0.10.0\scripts" -f "C:/Users/Blaise Jarrett/Documents/private_repots/blinky/openocd.cfg" -c "program \"C:/Users/Blaise Jarrett/Documents/private_repots/blinky/cmake-build-debug/i2c_display_example/i2c_display_example.elf\";reset;shutdown"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override u
se 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results mi
ght differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.230526
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080012d4 msp: 0x20005000
** Programming Started **
auto erase enabled
Error: Invalid command argument
image.base_address option value ('Jarrett/Documents/private_repots/blinky/cmake-
build-debug/i2c_display_example/i2c_display_example.elf') is not valid
embedded:startup.tcl:476: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 532
at file "embedded:startup.tcl", line 476

Process finished with exit code 1
`

elmot, 2018-03-27:
Looks like it won't work at all. Please do not use intl characters or spaces.

PR #93 (2018-03-27): Prohibit intl characters and spaces for folder/project names
Resolves #80

---
This change is [[image]](https://reviewable.io/reviews/elmot/clion-embedded-arm/93)
commit 0302dfef (2018-03-27): Prohibit intl characters and spaces for folder/project names (#93)

* Prohibit intl characters and spaces for folder/project names.

Resolves #80
closed 2018-03-27 by elmot