elmot/clion-embedded-arm #82: Wrong GDB-executable is executed.

Rumchiller, 2018-02-02:
## Expected Behavior
After starting the debug session, OpenOCD should flash the device and the selected GDB-executable should start up.

## Actual Behavior
After starting the debug session, OpenOCD starts flashing the device but the wrong GDB-executable is executed. Maybe it's related to the toolchain-settings. In my case the MinGW toolchain is selected, with modified C/C++ compilers and make, the debugger is MinGW. But inside the OpenOCD-settings the arm-none-eabi GDB-executable is selected.

## Steps to Reproduce the Problem

  1. Set under OpenOCD settings the GDB location to arm-none-eabi.
  2. Set under toolchain settings the debugger to MinGW.
  3. Start debug session.

## Specifications

  - Plugin Version: 1.0.beta3
  - CLion version: 2017.3.2 (173.4548.31) and 2018.1 (181.3263.17)
  - OpenOCD version: 0.10.0
  - GDB Version: from toolchain
  - STM32CubeMX version: none
  - Toolchain: GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git (arm-none-eabi)
  - Platform: Windows 10
  - Hardware configuration: STM32F429IG with ST-Link v2 standalone

elmot, 2018-02-03:
This is expected behaviour. The plugin has it's own debugger location parameter. Most probanly it will be changed in future versions.

closed 2018-02-03 by elmot

Rumchiller, 2018-02-03:
Yeah, that's the problem, sorry I didn't clarify that in my issue. The plugin starts the GDB-exeucutable which is defined in the toolchain settings, but as you mentioned, it should start the GDB-executable defined in the OpenOCD settings (this is also written into the workspace.xml file).