elmot/clion-embedded-arm #86: Fails on Debug

golddragon007, 2018-02-24:
## Expected Behavior
When I click on the Debug button it will compile, download and than goes into the debug session. This last one doesn't happens.

## Actual Behavior
After the build was successful I get this in the console:
Error during pretty printers setup: Undefined info command: "pretty-printer".  Try "help info".

Some features and performance optimizations will not be available.

info pretty-printer
Undefined info command: "pretty-printer".  Try "help info".

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.

Process finished with exit code 0

Before the "Process finished with exit code 0" line, I get a popup with successful firmware download.

## Steps to Reproduce the Problem

  1. Download this project: https://github.com/golddragon007/Copter-Core
  2. Open it and run it.

## Specifications

  - Plugin Version: 1.0.beta3
  - CLion version: 2017.3.3
  - OpenOCD version: 0.10.0
  - GDB Version: 7.8 mingw GDB (?)
  - STM32CubeMX version: 4.22.1
  - Toolchain: curl-7.54.1-win64-mingw
  - Platform: Windows 10 x64
  - Hardware configuration: STM32F4_DISCOVERY

golddragon007, 2018-02-24:
Here are some of my settings, maybe that's the wrong...  
OpenOCD:
[OpenOCD](https://ibb.co/bNnQXc)
Toolchain:
[Toolchain](https://ibb.co/giNsCc)
Cmake:
[cmake](https://ibb.co/hfnaXc)
Debug:
[debug](https://ibb.co/hJa7dH)

golddragon007, 2018-02-25:
gdb(alt-5):
Error during pretty printers setup: Undefined info command: "pretty-printer".  Try "help info".

Some features and performance optimizations will not be available.

info pretty-printer
Undefined info command: "pretty-printer".  Try "help info".

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.

Process finished with exit code 0

openocd(Alt-4):
"C:\Program Files\openocd-0.10.0\bin\openocd.exe" -c "tcl_port disabled" -s "C:\Program Files\openocd-0.10.0\scripts" -f board/stm32f4discovery.cfg -c "program \"D:/Orara/CLion/Copter-core/build/Copter-core.elf\";reset init;"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results mi
ght differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v27 API v2 SWIM v0 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 2.910729
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08005efc msp: 0x20020000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
Info : device id = 0x10016413
Info : flash size = 1024kbytes
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x20000046 msp: 0x20020000
wrote 32768 bytes from file D:/Orara/CLion/Copter-core/build/Copter-core.elf in
1.121366s (28.537 KiB/s)
** Programming Finished **
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000aa8 msp: 0x20020000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
adapter speed: 4000 kHz
Info : tcl server disabled
Info : accepting 'gdb' connection on tcp/3333
Info : dropped 'gdb' connection

clementnuss, 2018-03-07:
I have the exactly same issue. Any update ?

elmot, 2018-03-27:
Duplicate of #89.

Should be fixed in 1.0.beta4. Please do not use bundled GDB under windows!

elmot, 2018-03-27:
UPD: Bundled GDB should work again