elmot/clion-embedded-arm #89: Truncated register 16 in remote 'g' packet

leech001, 2018-03-08:
## Expected Behavior
Truncated register 16 in remote 'g' packet

## Actual Behavior
Truncated register 16 in remote 'g' packet

Process finished with exit code 0
Debugger disconnected

## Steps to Reproduce the Problem

  1. Run Debug

## Specifications

  - Plugin Version:
  - CLion version 2017.3.4:
  - OpenOCD version Version: 1.0.beta3:
  - GDB Version Open On-Chip Debugger 0.10.0 (2017-08-21):
  - STM32CubeMX version 4.24:
  - Toolchain MInGW:
  - Platform WIN10:
  - Hardware configuration: stm32f103c8t6

elmot, 2018-03-27:
Could you, please provide full logs from both, openocd and gdb?

elmot, 2018-03-27:
Should be fixed in 1.0.beta4. Please do not use bundled GDB under windows!

elmot, 2018-03-27:
UPD: Bundled GDB should work again

leech001, 2018-03-28:
Perfectly. Thank you very much !