elmot/clion-embedded-arm #91: CubeMX Makefile code generation support

phaethon, 2018-03-18:
## Expected Behavior
Latest CubeMX versions allow to select Makefile as a toolchain for code generation. Output is cleaner than SW4STM32 for the purposes of CLion import. Basically, only CMakeLists.txt has to be created.
Initial project import should work with Makefile output of CubeMX.

## Actual Behavior
Plugin does not create the CMakeLists.txt when importing Makefile code export from CLion.

## Steps to Reproduce the Problem
1. Choose Makefile as toolchain in CubeMX. 
2. Import project into CLion with OpenOCD plugin enabled.

## Specifications

  - Plugin Version: 1.0b3
  - CLion version: 2017.3.4
  - OpenOCD version:
  - GDB Version:
  - STM32CubeMX version: 4.25 (was the same in previous CubeMX version, too)
  - Toolchain:
  - Platform:
  - Hardware configuration:

elmot, 2018-03-28:
Last time when I tried that makefile it did not work well at least on Windows.
CLion 2018 is going to support makefiles, I will check out how it works at some point.

phaethon, 2018-03-28:
I am using the CubeMX generated Makefile on Linux and MacOS. Works fine for me. At present, I am manually converting it to cmake to use with CLion. Give me a notice if you want a working sample.

phaethon, 2018-03-28:
There is also custom target way:
https://stackoverflow.com/questions/26918459/using-local-makefile-for-clion-instead-of-cmake

elmot, 2018-03-28:
Well, the feature request looks more like "Support multiple build files generated from templates"

elmot, 2019-03-25:
This is not going to be implemented in open-source plugin. If the issue is still valid, please [create a feature request at JetBrains YouTrack system](https://youtrack.jetbrains.com/newIssue?project=CPP&c=Type%20Task&c=Subsystem%20Embedded). Please do not forget to provide an example.

closed 2019-03-25 by elmot