elmot/clion-embedded-arm #92: Feature Request: Pass gdb user arguments

kiciek, 2018-03-20:
I couldn't find a way to pass gdb arguments like pointing init scripts "---command=semihost.gdb" or any other way to tell gdb to run "monitor arm semihosting enable" at start.
I cannot write arguments in Settings->OprnOCD Support->GDB Location and I see no field for them.

elmot, 2018-03-28:
Unfortunately there is no such a place for the moment.
As a workaround, try to use your own OpenOCD board file and add `arm semihosting enable` there

elmot, 2018-05-03:
First of all, the feature is really nice, and I am missing you.

At the same time  it looks like CLion issue rather than my plugin issue, and I am not going to compete with them. There is an issue already https://youtrack.jetbrains.com/issue/CPP-7322

If somebody needs the feature, please try a workaround with CMake variable `CMAKE_C_FLAGS_DEBUG` similar as shown [here](https://intellij-support.jetbrains.com/hc/en-us/community/posts/206606295-Debugger-variables-issues?page=1#community_comment_206006829)

nicknolt, 2018-10-24:
Hi, i'm interested to have an example of cfg file modified.
By example in the nucleo f4 cfg file where to add the semihosting command? : 
```
source [find interface/stlink-v2-1.cfg]
transport select hla_swd
source [find target/stm32f4x.cfg]
reset_config srst_only
```

Thank you!

elmot, 2018-10-25:
My working example for F3-Discovery
```
# This is an STM32F3 discovery board with a single STM32F303VCT6 chip.
# http://www.st.com/internet/evalboard/product/254044.jsp

source [find interface/stlink.cfg]

transport select hla_swd

source [find target/stm32f3x.cfg]

reset_config srst_only
init
arm semihosting enable
```
Now semihosting output is visible in openocd window

nicknolt, 2018-10-28:
Thanks!
I will try this workaround  tuesday.

elmot, 2019-03-25:
Duplicate of https://youtrack.jetbrains.com/issue/CPP-9541

closed 2019-03-25 by elmot