elmot/clion-embedded-arm #103: "Update CMake project with STM32CubeMX project" throws error

AxxlForce, 2018-05-30:
After creating the project with STM32CubeMX and switching to CLion and hitting **Tools -> Update CMake project with STM32CubeMX project** throws an error. I don't know if it did what it was intended to do but it wont compile. Compile errors look like: **Error: unknown pseudo-op: ...**

## Steps to Reproduce the Problem

  1. Create project with STM32CubeMX
  2. hit **Tools -> Update CMake project with STM32CubeMX project**

## Specifications

  - Plugin Version: 1.1.alpha5
  - CLion version: 2018.1.3
  - OpenOCD version: 0.10.0
  - GDB Version: 8.1
  - STM32CubeMX version: 4.25.1
  - Toolchain: default (Ubuntu)
  - Platform: Ubuntu 18.04 x64

elmot, 2018-05-31:
@AxxlForce, not enough information. Please share your project
@dirtcrusher , try to  upgrade 2018.2 EAP + latest plugin (alpha6)

woniulive, 2018-06-03:
@elmot it seems that cMakeWorkspace.waitForReloadsToFinish(1000); get error in CLion version: 2018.1.3,just change it to cMakeWorkspace.waitForReloadsToFinish(); fix it.
[image]

elmot, 2018-06-03:
@woniulive , this method was changed recently, it's better to upgrade to CLION 2018.2 EAP
Can you, please, publish your CLion version? I will try to update the plugin metadata to avoid those incompatibilities.

woniulive, 2018-06-03:
@elmot i use Version: 2018.1.3,Build: 181.5087.36. it appears after i update to this version but it worked fine in previous version.

AxxlForce, 2018-06-04:
jap, updating to 2018.2 fixed it for me too!

AxxlForce, 2018-06-04:
ok now I get 

`/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/crt0.o: Conflicting CPU architectures 13/1`

after linking

mxcd, 2018-06-04:
> Conflicting CPU architectures 13/1
Getting the same error since switching to Ubuntu 18.04 LTS
Maybe 18.04 is using a newer (incompatible) version of the arm-none-eabi ?

elmot, 2018-06-05:
@AxxlForce , @mxcd , I can not reproduce that
I have Ubuntu 18.04 as well, toolchain 7.2.1
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
```

elmot, 2018-06-05:
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
Noted in the [USAGE.md](https://github.com/elmot/clion-embedded-arm/blob/master/USAGE.md#prerequisites)

The wrong toolchain trouble does not seem related to the plugin, I am closing the ticket.

closed 2018-06-05 by elmot

AxxlForce, 2018-06-06:
as @mxcd mentioned it's always a pain, if you're not able to use the packaged version. i followed your link and downloaded the toolchain from arm.com. unfortunately there is no instruction at all on how to install it... sigh. so i copied the content to /opt/ and added the bin folder to /etc/environment. still doesn't work:

```
/opt/clion/bin/cmake/bin/cmake --build /home/alex/Downloads/f4-testing3/f4-testing/cmake-build-debug --target f4-testing.elf -- -j 2
-- The ASM compiler identification is unknown
-- Found assembler: /usr/bin/arm-none-eabi-gcc
CMake Error at CMakeLists.txt:32 (PROJECT):
  The CMAKE_ASM_COMPILER:

    /usr/bin/arm-none-eabi-gcc

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/alex/Downloads/f4-testing3/f4-testing/cmake-build-debug/CMakeFiles/CMakeOutput.log".
See also "/home/alex/Downloads/f4-testing3/f4-testing/cmake-build-debug/CMakeFiles/CMakeError.log".
Makefile:710: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
```

elmot, 2018-06-06:
I added a ticket to make that automatable

mxcd, 2018-06-06:
@AxxlForce Just for reference:
In order to install the arm-none-eabi-gcc, I downloaded the desired version, made sure that the distro package is uninstalled (sudo apt purge gcc-arm-none-eabi), extracted the download in an folder of my liking and edited /etc/environment, adding this very folder to my path. Restarting the computer helps as well as deleting the "cmake-build-debug" folder. Cheers