elmot/clion-embedded-arm #105: get library config from .cproject

woniulive, 2018-06-13:
## Expected Behavior
After configuring library in sw4stm,clion can configure  library by reading .cproject.which means lines below can work:
<option id="gnu.c.link.option.libs.1978130072" name="Libraries (-l)" superClass="gnu.c.link.option.libs" useByScannerDiscovery="false" valueType="libs">
<listOptionValue builtIn="false" value=":STM32746G_Discovery_STemWin_Addons_GCC.a"/>
<listOptionValue builtIn="false" value=":STemWin540_CM7_OS_GCC_ot.a"/>
</option>

## Actual Behavior

## Steps to Reproduce the Problem

  1.configure library in sw4stm
  2.click update cmake project to ...
  3.library confuguration don't add to cmakelists.txt

## Specifications

  - Plugin Version:
  - CLion version:
  - OpenOCD version:
  - GDB Version:
  - STM32CubeMX version:
  - Toolchain:
  - Platform:
  - Hardware configuration:

elmot, 2018-06-13:
-  Please specify stm32cubemx version (you can find it in help->about menu)
- Please share, if possible, your .ioc file

woniulive, 2018-06-13:
@elmot it is added by sw4stm,not stm32cubemx.after exporting from stm32cubemx,import to sw4stm32,then it shows the code i uploaded.

elmot, 2018-06-14:
Sorry, I am afraid it's not going to be implemented in any visible future.

closed 2019-03-25 by elmot