elmot/clion-embedded-arm #108: Determine board config file from STM32CubeMX

csann, 2018-07-12:
## Expected Behavior
It would be very helpful if you were able to determine the board configuration file based on the STM32CubeMX project that was imported. If this were done, the projects debug configuration would typically require no user changes.

## Actual Behavior
The board config file must be set by the user even though the STM32CubeMX project already knows what board is being used.

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Plugin Version:
  - CLion version:
  - OpenOCD version:
  - GDB Version:
  - STM32CubeMX version:
  - Toolchain:
  - Platform:
  - Hardware configuration:

elmot, 2018-07-12:
The feature is partly implemented already, please use CLion 2018.2 EAP and the plugin 1.1.alpha7. The plugin tries to make a guess which file to use during project import.

Unfortunately it's hardly possible to keep the list of boards in sync with OpenOCD, because there are so many boards and openocd is updated every now and then.

Duplicate #54

closed 2018-07-12 by elmot

elmot, 2018-07-13:
[image]
It's here

csann, 2018-07-13:
Yes, I know of that. I’ve also had CLion pop up a dialog box...during importing, it think. Doesn’t happen all the time. 

I’m so happy I don’t have to use Eclipse anymore. Thank you for providing this plugin,