elmot/clion-embedded-arm #109: Path to elf file is not correctly escaped

lesto, 2018-07-16:
Host machine is windows but the build happen on ubuntu (WSL)

## Expected Behavior
The elf file get used for upload, from folder "c:/git projects/test/build/test.elf"

## Actual Behavior
image.base_address option value ('projects/test/build/test.elf') is not valid

## Steps to Reproduce the Problem

  1. move the project into a folder with spaces into the path
  1. try to run a openocd upload configuration

## Specifications

  - Plugin Version: 1.1alpha-4
  - CLion version: 2018.1.6
  - OpenOCD version: 0.10.0 (2017-08-21)
  - GDB Version: 7.10-1ubuntu3+9 (WSL)
  - STM32CubeMX version: none
  - Toolchain: arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3
  - Platform: windows + WSL

Radyl, 2018-10-23:
Having a similar issue with the linker here:

> AR19DD\~1.EXE: error: Projects/CLionTest/STM32F407VETx_FLASH.ld: No such file or directory
AR19DD\~1.EXE: error: Projects/CLionTest/cmake-build-debug/CLionTest.map: No such file or directory
mingw32-make.exe[3]: *** [CLionTest.elf] Error 1
CMakeFiles\CLionTest.elf.dir\build.make:974: recipe for target 'CLionTest.elf' failed

while the actual folder name is "STM Projects".
Will this be fixed soon?

elmot, 2018-10-24:
I am afraid not. Please rename the folder.

Radyl, 2018-10-27:
Good work anyhow!

elmot, 2019-03-25:
Duplicate of https://youtrack.jetbrains.com/issue/CPP-15289

closed 2019-03-25 by elmot