elmot/clion-embedded-arm #112: Unable to debug/launch if linker script has two applications

tmauron, 2018-07-24:
## Expected Behavior
Ability to debug an app that does not begin at `0x08000000` but at `ORIGIN = 0x08008000` for example. (With `VECT_TAB_OFFSET = 0x8000`). It works fine in the command line (openocd & gdb).
This is usefull if you have two applications. For example a bootloader and an application.

## Actual Behavior
FW download works (no error). When I press RESET the execution is stuck in the `disassembly` file.

## Steps to Reproduce the Problem

  1. Change linker script: `ORIGIN = 0x08008000` and `VECT_TAB_OFFSET = 0x8000`
  2. Try with command line (openocd & gdb) => this works. 
  3. Run with CLion debug (code stuck like there were breakpoints in assembly). Make sure there are NO breakpoints anywhere. (Same behaviour with breakponits).

Thank you!

## Specifications

  - Plugin Version: 1.1.alpha4
  - CLion version: 2018.1
  - OpenOCD version: 0.10.0
  - GDB Version: 7-2018-q2-update
  - STM32CubeMX version: 
  - Toolchain: 
  - Platform: MacOS High Sierra
  - Hardware configuration: stm32f4discovery

elmot, 2018-09-24:
Can you provide a reproducible example? 
is it possible to just flash a stub of the bootloader which just jumps to the correct address?

elmot, 2019-03-25:
This is not going to be implemented in open-source plugin. If the issue is still valid, please [create a feature request at JetBrains YouTrack system](https://youtrack.jetbrains.com/newIssue?project=CPP&c=Type%20Task&c=Subsystem%20Embedded). Please do not forget to provide an example.

closed 2019-03-25 by elmot