elmot/clion-embedded-arm #117: Document how to switch to a C++ Project

johnthagen, 2018-08-28:
It would be helpful if it were documented what needed to be done to switch to a C++ program.

From what I've read, renaming `main.c` to `main.cpp` is important, but beyond that I'm unsure what additional CMake changes need to be made to get everything (especially the STM32 HAL) to play nicely together.

For example, after making this change I seem to get linker issues building for the STM32F3DISCOVERY:

```
...
CMakeFiles/Project.elf.dir/startup/startup_stm32f303xc.s.obj: In function `LoopFillZerobss':
Project/startup/startup_stm32f303xc.s:111: undefined reference to `main'
```

elmot, 2018-09-02:
@johnthagen thanks for the idea
[Here is a small but working example](https://github.com/elmot/stm32f3-cpp-example/tree/master/Core/Src)
Documentation is not written yet.
Note the major part of the project is in C, and [some classes are used to control LEDs](https://github.com/elmot/stm32f3-cpp-example/blob/master/Core/Inc/cppest.h).

johnthagen, 2018-09-02:
@elmot Thanks! What do you think of #118? It seems to be working for me so far.

PR #118 (2018-09-04): Document how to change to a C++ Project
Fixes #117

---
This change is [[image]](https://reviewable.io/reviews/elmot/clion-embedded-arm/118)
closed 2018-09-04 by elmot