elmot/clion-embedded-arm #122: Improve WSL support

wschaeffer, 2018-09-24:
Host machine is windows but the build happen on ubuntu (WSL). Debugger should use windows path to .elf file. Instead uses WSL path to .elf file.

## Expected Behavior
Debugger is running on windows host.

## Actual Behavior
Error during pretty printers setup: Error while executing Python code.

Some features and performance optimizations will not be available.

Cannot load symbol file: /mnt/c/Users/ws/Documents/STM32/FLASH/cmake-build-debug/FLASH.elf: No such file or directory.

## Steps to Reproduce the Problem

  1. Try to debug program

## Specifications

  - Plugin Version: 1.1.alpha7
  - CLion version: 2018.2.4
  - OpenOCD version: 0.10.0-8 (2018-05-12)
  - GDB Version: Debian 7.12-6
  - STM32CubeMX version: 4.26.1
  - Toolchain: arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1
  - Platform: Windows 10 1803 + WSL (Debain)
  - Hardware configuration: STM32L072

elmot, 2018-09-24:
Try to manually set the binary file path in the run configuration.

wschaeffer, 2018-09-24:
Same error message, unfortunately. However, windows path is used when uploading. CLion runs:
`C:\Users\ws\Documents\OpenOCD\OpenOCD\0.10.0-8-20180512-1921\bin\openocd.exe -c "tcl_port disabled" -s C:\Users\ws\Documents\OpenOCD\OpenOCD\0.10.0-8-20180512-1921\scripts -f board/stm32l0discovery.cfg -c "program \"C:/Users/ws/Documents/STM32/FLASH/cmake-build-debug/FLASH.elf\"" -c reset -c shutdown`
Which surprised me.

JulianSchneider, 2019-03-14:
I'm having the same issue. I tried changing the 'exectuable' path directly over the windows path which didnt change anything. Also when chaning the openOCD GDB settings from 'shipped with clion' to 'from toolchain' the error goes away but the Debugging doesnt work. Is there anywhere else where I can set the windows path?

elmot, 2019-03-25:
Duplicate of https://youtrack.jetbrains.com/issue/CPP-15490

closed 2019-03-25 by elmot