elmot/clion-embedded-arm #123: Cannot load settings from file workspace.xml

pmjobin, 2018-09-26:
## Expected Behavior

## Actual Behavior

After having setup OpenOCD in the "Run/Debug Configurations" dialog and closing the project, opening the project again in CLion gives the following error message:

[image]

## Steps to Reproduce the Problem

1. Create a run/debug configuration for OpenOCD
2. Close the project to overwrite workspace.xml
3. Open the project again to see the error message

Here is the faulty line in the workspace.xml file as pointed out by the error message:

`<configuration name="NucleoH7.elf" type="elmot.embedded.openocd.conf.type" factoryName="elmot.embedded.openocd.conf.factory" PASS_PARENT_ENVS_2="true" PROJECT_NAME="NucleoH7" TARGET_NAME="NucleoH7.elf" CONFIG_NAME="Debug" RUN_TARGET_PROJECT_NAME="NucleoH7" RUN_TARGET_NAME="NucleoH7.elf" elmot-ocd:gdb-port="3333" elmot-ocd:telnet-port="4444" elmot-ocd:board-config="board/st_nucleo_h743zi.cfg" elmot-ocd:reset-type="INIT" elmot-ocd:download-type="ALWAYS">
      <method v="2">
        <option name="com.jetbrains.cidr.execution.CidrBuildBeforeRunTaskProvider$BuildBeforeRunTask" enabled="true" />
      </method>
    </configuration>`

## Specifications

  - Plugin Version: v1.2.alpha1
  - CLion version: 2018.3 EAP

elmot, 2018-09-30:
Please delete the file and recreate the run configuration. I think it would help.

pmjobin, 2018-10-03:
Thanks for your help. I tried deleting the embedded .clion folder which contains the workspace.xml file and reopen the cmake file in order to reset every settings but to no avail. As soon as I add the OpenCD Run/Debug configuration (which is stored in workspace.xml), CLion complains it can no longer open the workspace anymore. It seems related to the xml attributes starting with `elmot-ocd:*` because after erasing those manually from the xml file, I can open the workspace without the error message box popping up.

Thanks,
PM

elmot, 2018-11-22:
After some experiments I realized I hardly can fix that in my code. This issue is just a side effect of some my refactorings. 

How to fix:
1. close clion
1. drop workspace file reported to be bad
1. run clion and open the project again

closed 2018-11-22 by elmot

pmahot, 2018-11-29:
Hi,

I did try CLion 2018.3 and this issue seems to be back.
I did drop the workspace file and the issue was still there.
Each time I fix the project, it is broken at next CLion startup.

It seems that the "elmot" namespace definition is missing in workspace.xml file.
Something like following definition should appear in the configuration element:
`xmlns:elmot-ocd="https://github.com/elmot/clion-embedded-arm"`
Modifiying configuration node like this fix the project, but it is rewritten by CLion in the corrupted form.
`<configuration xmlns:elmot-ocd="https://github.com/elmot/clion-embedded-arm" name="MyName.out" type="elmot.embedded.openocd.conf.type" factoryName="elmot.embedded.openocd.conf.factory" nameIsGenerated="true" ...`

I think that something like following code in the writeExternal method from OpenOcdConfiguration should fix this.
`element.addNamespaceDeclaration(NAMESPACE);`

Let me know if you do something about this.

Thanks,
Pierre