elmot/clion-embedded-arm #128: OpenOCD support dialog incomplete. No way of specifying board

xerydf, 2018-10-23:
## Expected Behavior
pressing Control + Alt + S
and choosing Build, Execution , Development -> openOCE Support
A dialog looking like shown in the blog. ie. something like this:
[image]

is expected to apear

## Actual Behavior

Instead the dialog looks like the:

[image]

## Steps to Reproduce the Problem
As described above or 
Trying to flash and debug the build binary by
  1. Press Shif-F9
The this box pops up:
[image]

  2. Clicking Settings button, I'm returned to the dialog w. missing input fields for board coanfig etcc.
[image]
  

## Specifications

  - Plugin Version:
1.1alpha7
[image]

  - CLion version:
2018.2.5
[image]

  - OpenOCD version:
0.10.0

  - GDB Version:
Clion buldled version: 8.1

  - STM32CubeMX version:
[image]

  - Toolchain:
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]

- MinGW:
5.0
[image]

It's an unofficial build downloaded via link from th eopenOCD page. I followed the suggestion sposted on the blog januarey 9th by Max copying the scripts from openOCD_home to openOCCD_home/scare/scripts. this didn't make any difference.

  - Platform:
win10 Pro 1803 - 64Bit

  - Hardware configuration:
i3-4360 @ 3.7GHz & 8GB RAM

[image]

xerydf, 2018-10-24:
I didn't specify the board. It's a STM32F4Discovery

elmot, 2018-10-24:
The plugin was advanced a little big since that blog post. Now all those settings are in `Run/Debug  Configuration` dialog.

closed 2018-10-24 by elmot