elmot/clion-embedded-arm #131: Load Project Setting Error after add OpenOCD configuration

MarkCL, 2018-12-04:
## Expected Behavior
Project Setting Load OK.

## Actual Behavior
[image]

## Steps to Reproduce the Problem

  1.  Run->Edit Configurations->Add New Configuration choose OpenOCD Download & Run
  2. Click Apply and OK
  3. Restart CLion
  4. Project Setting Load error.

## Specifications

  - Plugin Version:  1.2 alpha
  - CLion version:   2018.3
  - OpenOCD version: 0.10.0
  - GDB Version:    8.0.1
  - STM32CubeMX version:
  - Toolchain:         gcc-arm-none-eabi-7-2018-q2-update
  - Platform: Mac OSX 10.14
  - Hardware configuration:

pmahot, 2018-12-04:
Hi,

This looks like a Duplicate of #123 

Regards,
Pierre

elmot, 2018-12-04:
Duplicate #123

MarkCL, 2018-12-05:
Hi Elmot,

It is duplicate #123, but the issue is back in the latest version CLion 2018.3 and plugin 1.2 alpha 2.
Do you have plan to fix it?
Thanks in advance!

Best Regards,

Mark

elmot, 2018-12-05:
Yes, I am working on it
Duplicate https://youtrack.jetbrains.com/issue/CPP-14863

commit 7b74133a (2018-12-05): Get rid of namespaces

Fixes #131
closed 2018-12-05 by elmot