elmot/clion-embedded-arm #135: Debug fails with "Error running 'OCD test-program': Debugger executable is incorrect"

Scottapotamas, 2018-12-27:
## Expected Behavior

1. Press the "Debug OCD $projname" button in toolbar/ F9
2. Firmware builds, flashes to target, attaches debugger, debug tab shows breakpoints etc.

## Actual Behavior

Build+run works fine. Can't debug.

1. Firmware builds, flashes to target.
2. CLion errors with "Debugger executable is incorrect" as below

```
12:46 AM	Clean finished in 140 ms
12:46 AM	Build finished in 2 s 80 ms
12:46 AM	Error running 'OCD delta-control': Debugger executable is incorrect
```

OpenOCD Console output:

```bash
/usr/bin/openocd -c "tcl_port disabled" -s /usr/share/openocd/scripts -c "gdb_port 2333" -f "/home/scott/projects/zaphod-bot/software/delta-control/delta-control Debug.cfg" -c "program \"/home/scott/projects/zaphod-bot/software/delta-control/cmake-build-debug/delta-control.elf\"" -c "init;reset init;"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
cortex_m reset_config sysresetreq
Info : No device selected, using first device.
Info : J-Link V9 compiled Oct 25 2018 11:46:07
Info : Hardware version: 9.40
Info : VTarget = 3.346 V
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Info : STM32F429VETx.cpu: hardware has 6 breakpoints, 4 watchpoints
adapter speed: 2000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08011390 msp: 0x20030000
adapter speed: 8000 kHz
** Programming Started **
auto erase enabled
Info : device id = 0x20016419
Info : flash size = 512kbytes
Info : Single Bank 512 kiB STM32F42x/43x/469/479 found
wrote 131072 bytes from file /home/scott/projects/zaphod-bot/software/delta-control/cmake-build-debug/delta-control.elf in 3.621807s (35.341 KiB/s)
** Programming Finished **
adapter speed: 2000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08011390 msp: 0x20030000
adapter speed: 8000 kHz
Info : tcl server disabled

```
No output is printed to the GDB console, it doesn't get that far...

If I manually launch a GDB server and configure CLion to use the remote server it will debug correctly.

## Steps to Reproduce the Problem

  1. Configure the OCD plugin as described in _clion-for-embedded-development-part-ii_
  2. Used this config https://gist.github.com/Scottapotamas/08abed9fd0baacc911b5c0dd968f6645
  3. Other settings as in image (2333 is the port for the JLink changed from 3333) [image]
  4. Toolchain is setup to use the arm gdb, [image]

## Specifications

  - Plugin Version: `1.1alpha7`
  - CLion version: `2018.2.5 (Build #CL-182.4892.24, built on October 17, 2018)`
  - OpenOCD version: `0.10.0`
  - GDB Version: tested `arm-none-eabi-gdb 8.2.50`, `/usr/bin/gdb 8.2`. Bundled GDB isn't available.
  - STM32CubeMX version: `4.25.0, but most (all?) cube libs have been replaced.`
  - Toolchain: `arm-none-eabi-*    (8-2018-q4-major) 8.2.50.20181213-git`.
  - Platform: `Solus Linux 4.19.9-105.current`
  - Hardware configuration: 
     - STM32F429VE* on custom PCB. SW4 works fine etc.
     - Using JLink-EDU (HW 9.4, SW built Oct 2018) as probe.

elmot, 2018-12-27:
Everything is too old.
Actual CLion version is 2018.3, and corresponding plugin version is published.

Won't fix

closed 2018-12-27 by elmot

Scottapotamas, 2018-12-27:
I updated to 2018.3.2, with plugin version 1.2alpha3.

Seem to have things working properly now, key setting seemed to be the OpenOCD plugin setting for GDB to "From toolchain (arm-none-eabi-gdb).

Thanks for your work on the plugin!