elmot/clion-embedded-arm #138: Support multiple config file

bkg2k, 2019-02-11:
## Expected Behavior
Working with STM32F4 and STLink debug devices, I'm able to select more than one config file in Eclipse OpenOCD plugin:
```
-f interface/stlink-v2-1.cfg
-f target/stm32f4x.cfg
```

## Actual Behavior
Only one config file can be selected.

## Specifications

  - Plugin Version: 1.3.Alpha
  - CLion version: EAP 2019.1
  - OpenOCD version: 0.10.0-5-20171110-1117
  - GDB Version: 
  - STM32CubeMX version:
  - Toolchain:
  - Platform:
  - Hardware configuration:

elmot, 2019-02-11:
Duplicate #136

elmot, 2019-02-11:
You can write your own config file for your project and use that. The file is going to have only references to existing configs + optional tweaks. Here is an example, standard file for STM32F4-Discovery from OpenOCD distribution.

```
# This is an STM32F4 discovery board with a single STM32F407VGT6 chip.
# http://www.st.com/internet/evalboard/product/252419.jsp

source [find interface/stlink.cfg]

transport select hla_swd

# increase working area to 64KB
set WORKAREASIZE 0x10000

source [find target/stm32f4x.cfg]

reset_config srst_only

```

closed 2019-02-11 by elmot